speckleworks/SpeckleRhino #15: Rhino Receiver Bake with Sublayers

didimitrie, 2017-12-12:
When baking a receiver, if identifying the classic rhino `::` layer subdivision syntax, re-enact it as such and not in a flat list as it is currently.

fraguada, 2017-12-19:
@didimitrie What is going on here? https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoPlugin/src/Interop.cs#L78-L94

Do we need to listen to these things? Another option would be to check if objects are selected or not. If they are not selected, and we need them to be, we can run an object getter. Might be good to discuss what you had in mind here.

didimitrie, 2017-12-19:
We listen to these here for the initial object selection and on the add/remove objects operation on a sender. It displays real-time what you have selected and on what layer...

[image]

Is there a problem this is causing down the line?

fraguada, 2017-12-23:
Ok. I have something working. One question, should each sublayer level keep a Guid? At the moment, if I am not mistaken, only the Guid of the lower level layer is stored in the stream.

didimitrie, 2017-12-23:
I think it's quite a cosmetic thing at the end of the day (we don't sync layers based on guids, etc) - so we could cut short the hassle of nested guids, no?

fraguada, 2017-12-23:
True. Just that there was [this line](https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoPlugin/src/SpeckleRhinoReceiver.cs#L270) where the Id of the layer was being set.

didimitrie, 2017-12-23:
Hm. We can do the same stuff, ie separate the guids of layers by `::` and be done with it. Otherwise, even easier, just let rhino set that guid... and that's that. I might be missing something - i  have this strange feeling like i forgot something at home (wallet, or keys, or headphones...).

fraguada, 2018-01-25:
After testing, this one seems ok.

closed 2018-01-25 by fraguada