speckleworks/SpeckleRhino #29: Rhino Receiver Conduit tweaks

fraguada, 2017-12-14:
Give a little love to this...mesh colors, etc.

fraguada, 2017-12-14:
In R5 this might be annoying since Vertex Colors don't pay attention to Alpha: 
[image]
These will occlude other objects and look exactly as they would in a Shaded Display mode if they were baked. 

In Rhino 6, this is not an issue since we can add some alpha to the Vertex Colors. It would theoretically be possible to make a bitmap from the vertex colors and map it to the mesh, but this seems like overkill.

didimitrie, 2017-12-14:
defs overkill. we keep them as they are now, would say - you've anyway got `DrawBoxCorners` around them when you hover...

fraguada, 2017-12-14:
Closing then...watch your colors people!

closed 2017-12-14 by fraguada