speckleworks/SpeckleRhino #34: AppVeyor build artifacts

pearswj, 2017-12-15:
Zip up build artifacts so we can provide a link to download the latest successful build from appveyor.

See https://www.appveyor.com/docs/packaging-artifacts/

didimitrie, 2017-12-15:
Should (can) they be separated into gh plugins and rh plugins?

pearswj, 2017-12-15:
They can and perhaps they should! As a first pass I was just going to provide an easy way for people to get their hands on the latest build. Later we can think about creating plug-in packages/installers/etc.

pearswj, 2017-12-15:
Added a link to the README in 8207552. Grasshopper only for now. Ran out of time, driving back to Bath in a sec. But it's a start!

fraguada, 2017-12-19:
I feel separate packages will create an unnecessary maintenance vector and dup dlls can be avoided. I can investigate the appropriate packaging mechanism to keep everything together.

closed 2017-12-19 by pearswj