speckleworks/SpeckleRhino #49: Package .rhp, .dll s, and .gha for mass consumption.

fraguada, 2017-12-19:
Related to #34. Can we leverage .rhi / .macrhi for distributing SpeckleRhino?

fraguada, 2017-12-19:
Here are the locations where GH looks for Add-Ons (from David Rutten):

> GH tries to find its plugins in the following locations:
> 
> 1. In the installation Components folder, though ideally only the standard GHA files should be here.
> 2. In an %appdata% folder. This is the recommended place for 3rd party GHAs.
> 3. There's an xml file somewhere in %appdata% which contains a list of optional search paths. Some complicated thord party GHA developers modify this xml to point to a custom installation folder somewhere.
> 4. **All folders registered by Rhino plugins.**
> 
> The %appdata% component folder can also contain .ghlink files which are textfiles with a .ghlink extension that may contain additional search paths. I added this option so people don't have to screw about with my xml files.

I've tested (with a test sln) option 4 with Rhino 5 (x64) sr14 and Rhino 6 BETA and it works just fine. This means we should be able to leverage the Rhino Installer Engine to place the SpeckleRhino plugin into the appropriate place, the .rhp, .gha, and .dlls, resources, etc in the same place. GH will look for any .gha where the SpeckleRhino.rhp is, and load the Add-On.

didimitrie, 2017-12-19:
This sounds perfect then!

pearswj, 2017-12-19:
Sounds like a good approach. For reference, I added some zipping-up steps to the "after build" script (Build tab) and then configured the artifact. If you arrange the files in the correct structure first – in a `dist/` folder, say – then you might be able to use [AppVeyor's own artifact zipping](https://www.appveyor.com/docs/packaging-artifacts/) to keep things simple.

pearswj, 2018-01-02:
Just double checked the code that does no. 4 (above) and it's limited to loading GHAs from the same directory as the RHP. Because of this restriction, we should try to avoid having the Rhino 5 and 6 plug-ins in separate subdirectories of the RHI package. Do these projects share the same dependencies/resources? (@fraguada – I'm thinking along the same lines as the Iris RHI, with the V5 RHP and V6 RHP side-by-side.)

fraguada, 2018-01-02:
> Do these projects share the same dependencies/resources?

@pearswj yes they do. Packaging like Iris for Windows is what I was thinking too.

fraguada, 2018-01-02:
Few things left to do for this one:

- [ ] Static UI dist and targeting in the UserControl
- [ ] zipping into an RHI
- [ ] Testing that
- [ ] CI stuff related to the same

fraguada, 2018-01-23:
https://github.com/speckleworks/SpeckleRhino/commit/4055b57a910835a58a66f7f0cb830f188dda4490 this fixes the post build events and adds things to a Debug and Release folder in the Solution Directory. NOTE: this hasn't been updated in CI yet, so those folder will likely not exist.

fraguada, 2018-01-23:
So this is what I have cooked up this far:
**Local Dev / Testing:** building `SpeckleWinR5`, `SpeckleWinR6`, `SpeckleGrasshopper`, or the entire solution copies resulting files to `$(SolutionDir)\$(ConfigurationName)`. This is useful for testing during development.
**CI Artifacts:** After building the solution release configuration, there is a post-build set of commands that `7z` the `SpeckleWinR5`, `SpeckleWinR6`, and `SpeckleGrasshopper` `bin\Release` folders into a specklerhino.rhi.

I still have one box to tick, which is the testing of all of this and making sure that this installs properly with the Rhino Installer Engine, and works with the built RhinoView app.

@didimitrie @pearswj does this sound reasonable?

pearswj, 2018-01-24:
Totally reasonable!

fraguada, 2018-01-24:
@pearswj @didimitrie there is one more rule, each project builds to their respective bin\$(ConfigurationName). I was about to kill this and dump everything into a solution wide Release/Debug directory, but I think it is good to keep the build output of each project separate, and automate the dump of those contents. Does this make sense?

didimitrie, 2018-01-24:
I believe it does. Not sure regarding implications, but from my point of view, this keeps things a bit more clear. 

During debugging, will we need to reference all dlls from inside the solution wide Debug or from the individual folders? Not a biggie, more for me to know what to expect re debugging in the near future...

fraguada, 2018-01-24:
This depends on the dev, but should not matter. The dev can debug from the rhp/gha in `$(ProjectDir)\bin\$(ConfigurationName)` or from `$(SolutionDir)\$(ConfigurationName)`. The only reason I can think of keeping build output in `$(ProjectDir)\bin\$(ConfigurationName)` is to keep a clean reference to that project's build output and dependencies.

pearswj, 2018-01-24:
I think how you have it now – leaving the individual `$(TargetDir)`s alone and carefully curating a central "distribution" folder – is a good approach.

fraguada, 2018-01-24:
Woo! Mass consumption packaging v1 complete.

closed 2018-01-24 by fraguada