speckleworks/SpeckleRhino #50: Exceptions thrown upon creating new Sender

fraguada, 2017-12-19:
https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoPlugin/src/Interop.cs#L401

Behavior: 
Whether an object is selected or not, when a new Sender is created, this line throws an exception.

fraguada, 2017-12-19:
Seems related to Rhino 6/BETA, perhaps to do with the layerIndex working differently...

didimitrie, 2017-12-19:
Nope. I'm on it. It's fixed in 6, testing 5. 
In R6, `SelectedObjects = RhinoDoc.ActiveDoc.Objects.GetSelectedObjects(false, false)` returns a list full of nulls if nothing is selected. In R5, it returns an empty list, which is why it was working before.

didimitrie, 2017-12-20:
This  was a messy one with things related to camelCaseing jsonconvert as well as other r6 vs 5 differences, and of course, mostly sloppy code by yours truly. I'm gonna close this one for now.

closed 2017-12-20 by didimitrie