speckleworks/SpeckleRhino #57: R6: Cef Init Twice

didimitrie, 2017-12-21:
Steps to reproduce:
- Open R6 with empty file
- Open Speckle Panel
- Open a different, existing file

didimitrie, 2017-12-21:
Been looking at this a bit. In R6, panels seem to get re-instantiated on new file & file open, resulting in cef being initialised more than once. This does not seem to happen in R5. 

Preventing that by 
```cs
if (!Cef.IsInitialized)
  Cef.Initialize(settings);
```
Does not solve the problem, as we still will initialise a _n_ load of browsers, where  _n_ = number of times a file was opened in that specific rhino instance. 

Only way I see forward is to adapt/modify the whole serialisation, etc. to work with the disposal mechanisms that we have currently triggered only on `RhinoApp.OnClose`. This will allow for a consistent approach through R5 and 6.

fraguada, 2018-01-22:
I doubt the plugin is reloaded. We could tie an instance of cef to the plugin instance. I'll try this tomorrow.

fraguada, 2018-01-25:
This commit https://github.com/speckleworks/SpeckleRhino/commit/d8a8151c93a9b759f8062ae984cca9be8f7cb853 fixes this issue.

closed 2018-01-25 by fraguada