speckleworks/SpeckleRhino #75: Packaging Issues

fraguada, 2018-01-23:
Getting some errors related to how the RHI is packaged, though I'm not surprised. The package itself should be tidied up. Should figure out how to keep CEF related stuff in a subdirectory. Related to #49.

- [ ] RHI is corrupt 
[image]
[specklerhino (20180123-235255).log](https://github.com/speckleworks/SpeckleRhino/files/1657911/specklerhino.20180123-235255.log)

- [ ] RHI GUID Mismatch
[image]
[Release (20180123-235127).log](https://github.com/speckleworks/SpeckleRhino/files/1657910/Release.20180123-235127.log)

fraguada, 2018-01-24:
https://github.com/cefsharp/CefSharp/pull/1753 this looks handy.

pearswj, 2018-01-24:
Eek. @fraguada shout if you need a hand!

didimitrie, 2018-01-24:
This looks really good  👍 even if it doesn't work!!! Sorry for the unhelpful comment...

pearswj, 2018-01-24:
@fraguada I think I've figured out why the zip archive is corrupted. `7z` must have been confused by the `.rhi` extension. Forcing `7z` to use ZIP compression seems to work. I also added the recursive flag for good measure (doesn't make a difference AFAICT, but the docs says otherwise!).

```
7z a -tzip -r specklerhino.rhi %APPVEYOR_BUILD_FOLDER%\SpeckleRhinoPlugin\SpeckleWinR6\bin\Release\*
7z a -tzip -r specklerhino.rhi %APPVEYOR_BUILD_FOLDER%\SpeckleRhinoPlugin\SpeckleWinR5\bin\Release\*
7z a -tzip -r specklerhino.rhi %APPVEYOR_BUILD_FOLDER%\SpeckleGrasshopper\bin\Release\*
```

Note also I've changed `*.*` to `*`, after reading the following ([source](https://sevenzip.osdn.jp/chm/cmdline/commands/add.htm)):

>_7-Zip doesn't use the system wildcard parser. 7-Zip doesn't follow the archaic rule by which *.* means any file. 7-Zip treats *.* as matching the name of any file that has an extension. To process all files, you must use a * wildcard._

pearswj, 2018-01-24:
And I guess all the Rhino plug-ins need to have the same GUID!

Yours,

Capitán Obvio

pearswj, 2018-01-24:
Just tested the release [here](https://ci.appveyor.com/project/SpeckleWorks/specklerhino/build/157/artifacts) and it installed properly for Rhino 5 and Rhino 6!

fraguada, 2018-01-24:
Artifact RHI installs, and initial tests firing up the plugin works well in Rhino 5 x64, Rhino 6, and GH in both.

closed 2018-01-24 by fraguada