speckleworks/SpeckleRhino #84: Browser isn't ready by the time Speckle is.

fraguada, 2018-01-25:


didimitrie, 2018-01-25:
Dev log: I think it's the events that Interop sets on the rhino `BeginSaveFile`, etc., potentially also touches on #85; they never get removed, leading to being fired three times, or four times, or _n times_, depending on how many file new/save operations you do...

Solution would be to keep the Interop obj, and just replace the browser inside it as 987d5d0 tries, when the panel gets reintialised. Nevertheless, this is not clean, so I think we'll need to have a dispose method on it.

didimitrie, 2018-01-25:
Log: 
`public SpeckleRhinoUserControl( )` is created every time R6 opens a new file. no idea how/where the old one is disposed of, but now forcing disposals on the Interop object in `BeginOpenDocument`.

Now R5 of course just crashes outright on initialisation...

commit cfa76399 (2018-01-25): fixes #84 and fixes #85 hopefully 😇
- move cef shutdown to plugin shutdown
- added method for store disposal (interop object).
- disposing of the store object on every document open
- initialisation of the panel now should not conflict with itself, since r6 decides to continuously reinit it.
closed 2018-01-26 by fraguada