speckleworks/SpeckleRhino #89: Check mixed data from Gh in Rh

didimitrie, 2018-01-26:
Check if on the same layer/stream we have a mixture of numbers and geometry whether the preview/rh plugins break.

didimitrie, 2018-01-27:
Seems fine. Though it's a bit confusing, as you don't get any indication of what objects are in there...

Will move this to 0.0.5.

fraguada, 2018-05-21:
How would this be useful in Rhino? I suppose seeing certain quantities would be interesting, but technically, there wouldn't be any way to wire these quantities to any operation. I'd `won't fix` this or push to a future target when there might be a clearer way to make use of these kinds of streams.

didimitrie, 2018-05-21:
Not about usage in rhino, more about getting what looks like an empty stream in  rhino and not being able to understand why. 

[image]

it's bad ui/ux.