speckleworks/SpeckleRhino #98: Receiver disconnects output node

dradomtsev, 2018-01-31:
I don't know is it a bug but when You connecting other ID to the same Receiver it disconnects output node
[image]

didimitrie, 2018-01-31:
Thanks for filing this! Unfortunately it's a pain to deal with how grasshopper handles creation of outputs, and how they're connected with the rest of the components. 

The  "A" input from the first stream is different than the "A" in the second one (by `guid` not by name!), therefore the disconnect. I'm always dong a diff by original input's guid, and this is how i keep track of what's going on, what needs to be added/replaced. I can't do it by names, as you can have theoretically to inputs/outputs named "A" on the same stream...

I'll leave this  one here, and label it as an enhancement, not bug 👍

didimitrie, 2018-03-30:
I'm going to  mark this as closed, as this is expected behaviour - guids differ, and i don't want to do a messy name based diff on layers, as it might cause more troubles down the line...

closed 2018-03-30 by didimitrie