speckleworks/SpeckleRhino #102: RH6 Breps and RH5 Receivers

didimitrie, 2018-02-01:
Send a brep from Rhino 6 to Rhino 5 - meep: `SerializationException: Unable to read ON_Object from binary archive`

@fraguada & @pearswj is this even possible to handle or is it going to be a won't fix label? 

Context: breps are serialised in a very lazy developer kind-of way (base64 encoded byte dump + a mesh equivalent).

fraguada, 2018-03-25:
@didimitrie I'm not sure this will be possible. I think the brep class changed some from R5 to R6. I can ask to get a better understanding, but might be a won't-fix.

fraguada, 2018-05-20:
I wonder if we could pass something to the UI [here](https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoConverter/src/SpeckleRhGhConverter.cs#L676) to turn the stream card red, or show that there is an error.

didimitrie, 2018-05-20:
Yep, we can display an error. That should be possible via `EventBus.$on( 'client-error', ( streamId, data ) => { ...`

Nevertheless, I'm not sure how to bubble up errors from the converter back to the Receiver, which can then do `Interop.NotifySpeckleFrame(...`. We'll have to look into this! 

In general i think this is a good idea, as there are always potential conversion errors.

didimitrie, 2019-06-05:
closing as Rhino 5 is no longer supported

closed 2019-06-05 by didimitrie