speckleworks/SpeckleRhino #113: TextEntity Preview with wrong size

didimitrie, 2018-02-09:
[image]
`Hi, code!` above: real size is the big one (above). When baking stream, all is well. When previewing, it defaults to a size of 1 unit :/

didimitrie, 2018-02-09:
RH5 preview clipped, though text is displayed at the correct size.
[image]

fraguada, 2018-02-28:
Not seeing the clipping in R5. Will keep investigating.

Getting a crash (with dmp) when Speckle starts converting R6 Annotations.
[image]
Whilst debugging, I don't get this.

In any case, the text height is properly recorded: https://s003.speckle.works/api/streams/BkyWdfNOf/objects/?omit=displayValue,base64 I made text with 10 as the height, and this is properly recorded in the object.

In SpeckleRhinoReceiver, line 240, the incoming object (with correct text height) gets Deserialized by the `SpeckleCore.Converter.Deserialize`. This method is still receiving the correct object attributes. It is actually in SpeckleCore.Converter.Deserialize line 163
`var result = methods[0].Invoke(o, new object[] { o });`
 that the properties get messed up. This could be for various reasons, but mostly because all annotations changed in R6, and SpeckleRhinoGHConverter uses Rhino 5 RhinoCommon.

didimitrie, 2018-02-28:
>It is actually in SpeckleCore.Converter.Deserialize line 163
var result = methods[0].Invoke(o, new object[] { o });
that the properties get messed up. 

This is actually weird. Does this mean we need to `#if R6` etc?

fraguada, 2018-03-01:
Maybe. In Iris I have an R5 and R6 ToJson projects, mostly for handling Annotations. Might have to do something like this and #if R6 as you say. I'll test it.

fraguada, 2018-03-01:
TextHeight cannot be set in RhinoCommon for R6. This seems to me to be a bug, and I've filed it. https://mcneel.myjetbrains.com/youtrack/issue/RH-44473
This is an issue now for calculating the bounding box, because an AnnotationBase object's TextHeight property cannot be set, thus the bounding box cannot be properly made around it.

fraguada, 2018-03-02:
The [RH-44473](https://mcneel.myjetbrains.com/youtrack/issue/RH-44473) bug has been fixed and will be available in R6.3. I'll continue to test with R6.3 to get the code ready for that release.

fraguada, 2018-05-22:
Update:
- R5: Clipping occurring as no way to get BB from a TextEntity that isn't in the Document. Text IS drawn at the right height. Worst case scenarios for fix:
  - add BB min/max pts to the Text Spec
  - Upon deserializing/receiving, explode text to curves and use that to compute BB.
- R6: Clipping occurring. Text is drawn at the right height.

The clipping problem would only occur if the stream only consists of text entities.

didimitrie, 2018-05-22:
Sounds like a complicated fix... inclined to plop `wontfix` on this one if it takes too much effort. Random thought: couldn't we invent a bbox by just looking at text size + length (ie, 36 chars on 3 lines at 3pt => +- of a bbox of xyz)?

didimitrie, 2019-08-06:
closing as it's 1yr+ of non-activity. cleanup!!!

closed 2019-08-06 by didimitrie