speckleworks/SpeckleRhino #115: Display conduit hover is slow and wrong colour

didimitrie, 2018-02-09:
And laggy. Check with @PaulPoinet re speeding up/removing boxes.  
Colour, according to @fraguada should be the "speckle blue" :)

didimitrie, 2018-03-25:
I'll pm you a big fat stream

fraguada, 2018-03-25:
But it seems the laggyness is on the SpeckleView side of things, since it is accumulating and eventually processing mouseover events?

didimitrie, 2018-03-25:
Yes indeed. I was just trying to capture the behaviour in a gif.

didimitrie, 2018-03-25:
I'll stop, send you a big fat stream,  and the last layers are annoying. It think it can be potentially  debounced on the viewer side so as to send less events over...

fraguada, 2018-03-25:
The color issue at least taken care of. @didimitrie mentioned we should debounce over events in [SpeckleView](https://github.com/speckleworks/SpeckleView/issues/2).

PaulPoinet, 2018-03-25:
Not sure if it's today's problem but as far as I remember the issue was with the box edge corners. With a big stream, it starts to get very slow to display them as you need to calculate the bounding box for each object of the stream. I would suggest to get completely rid of them if it's not the case already

didimitrie, 2018-04-01:
Somehow rhino sender clients now don't have the hover preview working. All of the above are solved (see speckleworks/SpeckleView#2)

didimitrie, 2018-05-24:
I think this issue is superseeded by #178, and what it was initially raised for got fixed.

closed 2018-05-24 by didimitrie