speckleworks/SpeckleRhino #130: Texture coordinates on meshes increase size by 30%

didimitrie, 2018-02-19:
Is there any way we could cut back on this? It's a rather high cost for something that's not that used (in my bubble of  the world). Would it be a possibility to ignore them on meshes that don't have them explicitly set? 

Or just move them to the `properties` part, and make them optional, with a default to false? Asking this as a question. Personally I would take them out and reintroduce them when the masses ask for them...

tsvilans, 2018-02-19:
The `properties` option seems to make sense. Then you could have a 'getTextureCoordinatesAsProperties' component / extension method or something. IMO better than flat-out removing them... it would seem to be a bit counter-Speckle to pick and choose which aspects of an object get serialized...

This size limit on payloads is annoying... is there any way that payloads larger than the limit can be hacked up into smaller ones? Maybe separate issue...

didimitrie, 2018-02-21:
yes, they can be hacked - and they are now, automatically. problem is still with _singular_ objects that hit that threshold...

it's annoying, but it would require too much work for now. I'm currently limiting single objects to 2e6 bytes before deflation. I could increase this...

https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoPlugin/src/SpeckleRhinoSender.cs#L326

didimitrie, 2018-02-21:
also related to #128

tsvilans, 2018-02-22:
If I get some free time, I could look into extracting Texture Coordinates as a separate property before they get turfed?

didimitrie, 2018-02-22:
in grasshopper you could attach them via the user dictionary components which #116 💥 and will also raise a new issue on how to attach a list (in gh only, but a comma separated string should work well too).

didimitrie, 2018-03-30:
I've added in the upcoming release some basic global settings for the converter, which include `public static bool AddMeshTextureCoordinates = false;`

tsvilans, 2018-05-08:
Also, I can contribute a script component I wrote that extracts tex coords from a mesh as a space-separated  string and converts it to base64, and back again. Adding this to an object via User Data seems to be the Specklish way. Once we have a place to stick these little additions (re: issue #167 ) I can share it.

didimitrie, 2018-05-18:
I'll close this one now. "Official" stance for now:  
- no mesh coordinates by default
- add them via user data and interpret at  the receiver end

closed 2018-05-18 by didimitrie