speckleworks/SpeckleRhino #133: UI Conflicts

didimitrie, 2018-02-21:
[image]
can't click on layer visibility

[image]
can't click on "delete"

🥇

fraguada, 2018-03-25:
The 'delete' issue seems to be solved as that card goes on top of the '+'. 
For the other issue, what about just having a footer, and the client buttons fly out to the left?

didimitrie, 2018-03-25:
i was thinking something really easy, just add a `speckle-client` class to all receiver/sender containers and then drop a css rule `.speckle-client:last-child{ margin-bottom:  1000000px}`

fraguada, 2018-03-25:
I has a similar thought. That would at least always keep some whitespace that can be scrolled to. Weird, but might work.

didimitrie, 2018-03-25:
simple, and easy. i  don't think necessarily "weird"

didimitrie, 2018-04-01:
fixed in speckleworks/SpeckleView#3

closed 2018-04-01 by didimitrie