speckleworks/SpeckleRhino #140: What to do about R6 Dimension Styles?

fraguada, 2018-03-01:
What would be the strategy to add support for Rhino 6 Dimension styles without polluting SpeckleCore with app specific stuff? I guess SpeckleAnnotation can have user data, so that might be the place to start? Then during deserialization, if R6, check UD and if there is a DimStyle entry, suck it up.  🤔

didimitrie, 2018-03-01:
What about creating a new SpeckleObject in Core that handles dimensions, ie `SpeckleDimension`? Would have SpeckleAnnotation for:
- text dots
- text

SpeckleDimension: 
- linear dim
- radial dims
- etc. 

Watcha think? It's wide used enough to enable this as a fundamental object?

fraguada, 2018-03-01:
DimensionStyles in R6 apply to all AnnotationBase objects, including TextEntity, lineardims, etc.

didimitrie, 2018-03-01:
so then it doesn't make sense to split them up? my lizard brain says yes, but then we could resolve this by adding `_type` prop to spk annotations that tells us more explicitly what it is, and then we just leave everything else for the `properties`/UD

fraguada, 2018-03-01:
Moving this to 0.6.0 for now. There should be basic Text / TextDot support in 0.5.0. Relates to #113.