speckleworks/SpeckleRhino #146: Allow manual refresh in Receivers

didimitrie, 2018-03-14:
- [ ] Grasshopper - implement as context menu option
- [ ] Rhino - see where to fit another button

fraguada, 2018-03-25:
This is more of a SpeckleViewer issue, but if stream is paused, perhaps then the manual refresh button shoulld appear.
[image]

fraguada, 2018-03-25:
Wait, we don't need this. Maybe I misunderstood, but when would this be needed in Rhino? Already in SpeckleView we get 'stream outdated' messages in receivers and senders. 🤦‍♂️ 
When would we need a manual refresh button in Rhino?

didimitrie, 2018-04-01:
i've added a force refresh context menu option to the grasshopper components; we're good. i'll close

commit 1b723177 (2018-03-30): fixes #146
closed 2018-04-01 by didimitrie