speckleworks/SpeckleRhino #147: Speckle panel does not load UI

Crashnorun, 2018-03-15:
I've cloned the SpeckleRhino repo. To resolve some error I unloaded the RhinoMac and Rhino6 projects. I can build the SpeckleRhino project with no errors. When I load the RHP into rhino, the Speckle panel displays but no UI elements are visible. 
[image]
I've opened the CMD and navigated to the SpeckleRhinoView folder. I've run _npm run dev_ and a series of errors have displayed.
[image]

Any thoughts what I may be missing?

didimitrie, 2018-03-15:
hey charlie! yes, an `npm install`

fraguada, 2018-03-16:
I've updated the [README.md](https://github.com/speckleworks/SpeckleRhino/blob/master/README.md) and made a page on the wiki regarding the [Setup, Build, and Running procedures](https://github.com/speckleworks/SpeckleRhino/wiki/Develop-SpeckleRhino).

closed 2018-03-17 by fraguada