speckleworks/SpeckleRhino #150: Debug targets missing on fresh clone

didimitrie, 2018-03-18:
I followed the instructions in https://github.com/speckleworks/SpeckleRhino/wiki/Develop-SpeckleRhino and after cleaning/rebuilding, etc. on starting a debug session i got an error that got fixed by specifying Rhino.exe as a Debug start action. 

It seems these settings were not carried over - should this maybe go in the wiki, or is it just a problem on my side?

fraguada, 2018-03-18:
I'll investigate.

fraguada, 2018-03-25:
Which configuration where you running? Debug|AnyCPU should have R5 or R6\Rhino.exe: [R5](https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoPlugin/SpeckleWinR5/SpeckleWinR5.csproj#L189), [R6](https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoPlugin/SpeckleWinR6/SpeckleWinR6.csproj#L160), [GH](https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleGrasshopper/SpeckleGrasshopper.csproj#L157) (gh seems to have a csproj.user file in src)

We exclude .user files from source, but Debug | AnyCPU should work on a fresh clone.

didimitrie, 2018-03-30:
i'll close this as it's probably due to my setup

closed 2018-03-30 by didimitrie