speckleworks/SpeckleRhino #159: Grasshopper New Account Registration Fails

didimitrie, 2018-03-30:
as reported by @tsvilans 
[image]

Coupled with this, I  can't seem to be able to debug the popup properly :(

TTWDmitryZelenkin, 2018-04-03:
Have the same problem when compile from sources.

didimitrie, 2018-04-03:
I would really want to get rid of this completely. Not sure how to go about it, ideas are welcome...

tsvilans, 2018-04-03:
Will try with Rhino 5 when I get some time... Haven't had the chance to look at the code for this.

radumg, 2018-04-12:
getting this with Rhino5 x64 too - have we tracked down where the issue is yet ? I could try to squash the bug if so.

didimitrie, 2018-04-21:
Figured it out. For one reason or the other, the popup was an ancient nuget reference, and not the project itself. 

This is now fixed, and updated to use 1.0.0, and the popup is redesigned.

commit 7c12595a (2018-04-21): fixes #159 (and redesigns the form + adds a login option)
closed 2018-04-21 by didimitrie