speckleworks/SpeckleRhino #161: Dimensions not displaying in viewer

webthethird, 2018-04-05:
Hello,

I installed SpeckleRhino 1.0.0 yesterday. I am using the viewer to stream a building model for planning an event, and it is important to include the venue's dimensions. Though the dimension objects appear under the stream's layers, they are invisible in the viewer. I have tried exploding the dimensions and extruding the lines to make solids, in case annotations are not supported yet, but they still do not appear in the viewer. Any suggestions?

Best regards,
Bill Bodell

[image]
[image]

webthethird, 2018-04-05:
Update: the dimensions displayed when I exploded the extrusion objects and then rejoined them. So _extrusion_ and _dimension_ objects do not work but _polysurfaces_ do.

fraguada, 2018-04-05:
Annotations/Dimensions are not yet supported in the viewer: https://github.com/speckleworks/SpeckleViewer/issues/16

Extrusions I'll have to check, but I thought they were ok? @didimitrie @mswaidan ?

didimitrie, 2018-04-05:
@webthethird: do extrusions work rhino-to-rhino? regarding the online viewer i'll leave it up to @mswaidan . 

In Rhino it was anyways a mess and I think there were some problems that still need to be nailed, as far as i can remember simple extrusions (no holes for example) was ok though. 

d

webthethird, 2018-04-05:
@didimitrie I haven't tried rhino-to-rhino, since I am the only person on my team who works with CAD. My primary interest in Speckle is the ability to share my model with collaborators online without needing Rhino on their end.

mswaidan, 2018-04-05:
@webthethird I'll look into adding extrusion support, thanks for reporting.

mswaidan, 2018-04-06:
@fraguada @didimitrie:  
In my testing, extrusions don't appear to be working Rhino to Rhino.

Couple of issues that will need resolved before we can add support in viewer:

1.  resources.profile only contains one curve. This should be an array containing all the curves defining the shape to be extruded (e.g. a rectangle with a hole in the middle - resources.profile only contains the rectangle)

2. XYZ values in resources.profile.value for points in the profile curve are flattened. Profile.value should be an array of [x,y,z] arrays.

```profile: 
closed: true
geometryHash: "cef1e4240124"
hash: "bd25b6a9f827a804d9b0a6a88e94bbbf"
type: "Polyline"
value: Array(15)
0: -6.733733079946926
1: -12.513361930683487
2: 0
3: 21.26085448894375
4: -12.513361930683487
5: 0
6: 21.26085448894375
7: 15.481225638207206
8: 0
9: -6.733733079946926
10: 15.481225638207206
11: 0
12: -6.733733079946926
13: -12.513361930683487
14: 0
```

didimitrie, 2018-04-06:
agree re 1). I'm  not sure how rhino ones really work, but matthew if you wanna get your hands dirty with some .net c# rhinocommon madness - please do, you'll save some of my sanity!!!

Re 2) i'm afraid  this how polylines are spec'ed out in the name of efficiency. Just do a for( ...;  i+=3) loop...

Changing  that would require a big overhaul in speckle core and specs, which i would stay away from   at almost all costs

didimitrie, 2018-10-25:
closing as this one's more for the [Speckle Viewer](https://github.com/speckleworks/SpeckleViewer).

closed 2018-10-25 by didimitrie