speckleworks/SpeckleRhino #163: Sender component should have at least one input when placed on the GH canvas

marcsyp, 2018-04-16:
#### Expected Behaviour
A sender should have an input when it is placed on the canvas.

#### Actual Behaviour
A new user will place a Sender component on the canvas and see the message bubble, click on it, and see the message "Data Sent".  Wait, what?  How?  The component has no inputs.  The user must figure out that they must zoom in to use the ZUI to add an input.  I would argue that there is never a reason to have a sender that doesn't have an input... although in this world of GH hackers you never know.  :)

#### Proposed Solution (if any)
Simply populate one input to the sender on instantiation.

didimitrie, 2018-04-17:
it actually comes out of my laziness to initialise a non empty stream, which was a challenge in previous api versions, now it's not, so it should be an easy fix to  initialise a sender with however many inputs you want...

marcsyp, 2018-04-17:
Haha -- I see what you did there.  You're trying to get me coding, not just shooting UX bullets from the peanut gallery...

commit 384c3844 (2018-04-21): fixes #163 while i'm at it
closed 2018-04-21 by didimitrie