speckleworks/SpeckleRhino #167: Feature proposal: Contrib components

tsvilans, 2018-05-03:
### Feature Proposal
Speckle's core functionality could be greatly extended with contributed components / scripts for convenience and automating user data creation and customization. While small and simple in many cases, they could be useful for many.

For example, I find texture coordinates helpful. Most people don't care. These make meshes heavy and are therefore ignored by default. I've written a component that instead adds texture coordinates to the Mesh's user data, so that I can pass it through Speckle anyway when I need it.

A repo or some sort of policy for a curated list of contributed Speckle extensions would help to keep these sorts of additions up-to-date and centralized.

pearswj, 2018-05-03:
Sounds like we could use a package manager 👀

didimitrie, 2018-05-03:
Hehe, i see what you did there  @pearswj! Yes, for sure - we should have all speckle rhino avaiable through `yak`; i just relied so far on your and  @fraguada rhi magics :) 

My question: should  these components be incorporated in here - and displayed in a "Speckle Extensions" tab in gh, or be independent? @tsvilans 's suggestion of a component is a good one, and i'd have that available out of the box. 

I'm inclined into centralising them here to be honest. But willing to hear opinions!

tsvilans, 2018-05-03:
Perhaps something like a Speckle Milkbox? A separate community-driven list?

fraguada, 2018-05-03:
I'd be interested in certain guidelines that would allow these contributions to be as abstract as possible so that that could be used across platforms.

I also like the idea of having a known source where these extensions are found and published.

tsvilans, 2018-05-03:
> I'd be interested in certain guidelines that would allow these contributions to be as abstract as possible so that that could be used across platforms.

Many of them will be software-specific though, to translate different data models to Speckle and back...

didimitrie, 2018-05-03:
>I'd be interested in certain guidelines that would allow these contributions to be as abstract as possible so that that could be used across platforms.

I think @tsvilans mostly has in mind grasshopper specific hacks! 

>Many of them will be software-specific though, to translate different data models to Speckle and back...

I like the milkbox idea. How would that be implemented? Repo? Besides that it should all use YAK, rhino's package manager to  distribute. 

But now this can be cool,  because from the same source we could get components to  actually create all sorts of custom data structures, which has a larger implication. 

I don't want to diverge this  though, so by all  means, a quick dirty repo will do for now if that's the way we go (and not integration in here).

didimitrie, 2019-01-12:
I'll close this and suggest that all such useful scripts or snippets can (and should) be:

1. linked from the readme if too lazy to do (2)
2. **submit a PR with some new components!**

(2) would be preferable.

closed 2019-01-12 by didimitrie