speckleworks/SpeckleRhino #170: ARC conversion wrong in some cases!

Eckart-S, 2018-05-10:
I tried to fix a very annoying issue with Arc conversion from RhinoCommon to DesignScript but I figured out that this seems to be a RhinoCommon To SpeckleCore issue instead!?

I use GH to send the following Rhino geometry (2 lines + 5 arcs) to Speckle Stream SJvCE4GAM but as you see the geometry generated from GH Receiver, RH Receiver and Dynamo Receiver is all wrong. The Arc does not start in the proper location so it could be an issue with a badly serialised plane? Most of them work but some cases (like this one) turn out bad...

[image]

[ArcTest.zip](https://github.com/speckleworks/SpeckleRhino/files/1993264/ArcTest.zip)

Can someone figure out what the problem is and if the plane is properly stored?
Lines + Arcs are the most important elements for Revit/Dynamo so not being able to trust their validity is a serious issue and should be fixed with high priority.

Thanks a lot & Best regards, Eckart

didimitrie, 2018-05-16:
potentially related to #174 but it doesn't look like it. To keep an 👁 out for both

didimitrie, 2018-05-20:
I've looked things up: the problem is that those arcs have a their domain, ie: `-1.041238 To 0
`, and speckle doesn't take domains into account. If you flip them, everything works as expected. 

Todo: 
https://github.com/speckleworks/SpeckleSpecs/issues/25
https://github.com/speckleworks/SpeckleCore/issues/71

Then implement this in the rhino converter