speckleworks/SpeckleRhino #172: Extrusions do not preview.

fraguada, 2018-05-14:
Step 0: 
- [x] I've  read the [contribution guidelines](https://github.com/speckleworks/SpeckleOrg/blob/master/CONTRIBUTING.md)!
#### Expected Behaviour
When receiving objects that originated as extrusions, be able to viz them in Rhino.

#### Actual Behaviour
Pure extrusion objects do not seem to actually be displayed.

#### Affected Projects
SpeckleRhino

#### Reproduction Steps & System Config (win, osx, web, etc.)
1. Create a box
2. Send the box via SPK
3. In a new file or other instance of Rhino, receive the box.
4 Box does not display

#### Proposed Solution (if any)
Check what is happening with DisplayConduit. This does not happen if the extrusion is exploded, or if it is just a surface/polysurface object.

fraguada, 2018-05-18:
Seems the problem starts here in the [SpeckleRhGhConverter](https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoConverter/src/SpeckleRhGhConverter.cs#L700-L734). When the `ToNative()` method looks at the extrusion, there are two properties for the `extrusion.Profile.Type`:

1. SpeckleAbstract
2. Polyline (in the case of a box)

If the debugger goes on, it chooses the first one, and returns a null profile. If I move the debugger into [line 714](https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleRhinoConverter/src/SpeckleRhGhConverter.cs#L714),  the profile is not null, and the extrusion is displayed (albeit incorrectly due in part to the closed curve issue raised by Toni O.

didimitrie, 2018-05-18:
> there are two properties for the extrusion.Profile.Type

This is really weird. A speckle object can have only one type, and it's a string! How can this happen?

fraguada, 2018-05-18:
Not sure. The object profile type is Polyline: https://hestia.speckle.works/api/v1/streams/BJJpIHnAM/objects/?omit=displayValue,base64

didimitrie, 2018-05-18:
Okay this is weird. I remember at one point overriding the `type` prop with a `override` in all the speckle objects[1]. Not  sure why this didn't pull through after  the 1.0.0 regen. It might be causing this, though this doesn't explain why other objects work fine  :/

[1] https://github.com/speckleworks/SpeckleCore/blob/c561756a9ce85dd6a8ce219806d7a889b19adfa7/SpeckleCore/BaseObjects.cs#L732

fraguada, 2018-05-20:
I recommend removing the `SpeckleObjectType Type` property from ModelObject and the same from any other object.

There are two cases (I think) where PolyCurves and Extrusion profiles have a switch case to test for type. There are a few potential workarounds, including `.GetType().ToString()` which is bad design. In C# 7 one can do pattern matching: https://stackoverflow.com/a/299001/2179399. 

In this particular example, using the following works:
```
switch ( extrusion.Profile )
      {
        case SpeckleCore.SpeckleCurve curve:
          profile = ( ( SpeckleCurve ) curve ).ToNative();
          break;
        case SpeckleCore.SpecklePolycurve polycurve:
          profile = ( ( ( SpecklePolycurve ) polycurve ).ToNative() );
          if ( !profile.IsClosed )
            profile.Reverse();
          break;
        case SpeckleCore.SpecklePolyline polyline:
          profile = ( ( SpecklePolyline ) polyline ).ToNative();
          if ( !profile.IsClosed )
            profile.Reverse();
          break;
        case SpeckleCore.SpeckleArc arc:
          profile = ( ( SpeckleArc ) arc ).ToNative();
          break;
        case SpeckleCore.SpeckleCircle circle:
          profile = ( ( SpeckleCircle ) circle ).ToNative();
          break;
        case SpeckleCore.SpeckleEllipse ellipse:
          profile = ( ( SpeckleEllipse ) ellipse ).ToNative();
          break;
        case SpeckleCore.SpeckleLine line:
          profile = ( ( SpeckleLine ) line ).ToNative();
          break;
        default:
          profile = null;
          break;
      }
```

didimitrie, 2018-05-20:
> I recommend removing the SpeckleObjectType Type property from ModelObject and the same from any other object.

We need that one way or another for deserialisation in core (from `json` to `SpeckleObjects`). Different name might make more sense to avoid confusion. It's essentially a customised (for more control, afaik - i haven't really touched that part of the code since 1 year or more, so i'm rusty) version of Json.NET's `$type`. 

I've gotten rather confused by all the `types` discussed. I've reproduced the above problem. I think though this has nothing to do with the name of the variable, it's more a problem that: 
- SpeckleExtrusions store their profile in a generic SpeckleObject (as it can be curves, polylines, etc) as we don't have an `ISpeckleCurveLikeThing` interface.
- The Profile property in a SpeckleExtrusion is properly deserialised to whatever it is - polyline, curve, etc. 
- Nevertheless, since it's kept in a SpeckleObject, its visible compile time `Type` is `Abstract`, which is the default for generic speckle objects. 

Hence, when you GetType() it you get `SpecklePolyline` and when you just get its `Type` property it's abstract, as in all honesty the compiler thinks it's a SpeckleObject. 

This is some bad OOP on my side, i'll follow up with some ways out.