speckleworks/SpeckleRhino #195: Add SpeckleRhino to Yak

fraguada, 2018-07-18:
@pearswj I'd like to revisit adding Speckle to Yak, and I recall, though not very well, that there were some roadblocks to do so. Did this have to do with the fact that we have one rhi for both Rhino and GH functionality? @didimitrie seemed to recall this might have been the issue. 

I suppose that one combined installer would trip up things built off of Yak, such as the GH Missing Libraries functionality? Can this install RHIs?

fraguada, 2018-07-19:
After discussing Yak behavior with @pearswj, it seems the main issue with these kinds of combined packages (rhp + gha) is that the rhp takes precendence. At the moment, the only functionality that we could benefit from is if a user opens a gh definition with some Speckle component, Yak would detect it and install SpeckleRhino. At the time of writing, Yak looks at the package and grabs the GUID of the rhp, if one is present, otherwise, the gha. This might change in the (near) future, and @pearswj has some ideas. Barring this, and for the short term, we could manually edit the yak package manifest to include the GUID of the gha, and not the rhp.

pearswj, 2018-07-19:
P.S. it's worth checking out the GhCanvasViewport project, in particular the [GhCanvasViewportInfo.cs](https://github.com/mcneel/GhCanvasViewport/blob/master/GhCanvasViewportInfo.cs) file. Note that the `Name` matches the name of the [package](http://yak.rhino3d.com/packages/GhCanvasViewport) and the `Version` is taken from the `AssemblyInformationalVersion` attribute which makes it easy to keep the Grasshopper version in check by updating the assembly version numbers (I used this project as an example of how you could push packages to Yak from AppVeyor (see [Automation and CI](https://discourse.mcneel.com/t/rhino-package-manager-update-july-2018/66298)).