speckleworks/SpeckleRhino #216: Small curves render coursely

mswaidan, 2018-11-18:
#### Expected Behaviour
Small curves and large curves should render similarly in the viewer

#### Actual Behaviour
Small curves look chunky

#### Affected Projects
SpeckleViewer

#### Reproduction Steps & System Config (win, osx, web, etc.)
Rhino5 & Rhino6 sender to SpeckleViewer

#### Proposed Solution (if any)
Adjust curve handling in converter.js

didimitrie, 2018-11-18:
Needs to be controlled in the way we generate the `displayValue` of the nurbs curves in the rhino converter.

didimitrie, 2019-08-06:
@mswaidan , is this still an issue? I don't remember 😬  I think yes, but is it *really* needed?

mswaidan, 2019-08-06:
@didimitrie  AFAIK, yes, but since I'm no longer using Speckle for small objects I haven't noticed it. I think we can close and address it if it comes again.

closed 2019-08-06 by mswaidan