speckleworks/SpeckleRhino #241: Rhino Plugin on Bake doesn´t keep user Data

e-lupo, 2019-03-04:
when baking items from a receiver using the Plugin within Rhino, User Data attached to geometry in Dynamo doesn´t transfer over to User Attributes in Rhino.

didimitrie, 2019-03-04:
thanks for flagging! to check and weigh what to do: 

- add another tab to the rhino plugin ui that displays user dictionaries
- get @PaulPoinet to finish up his thesis demonstrators in a way that it can be used in specke
- ping and beg mcneel to update the user data tab to display dictionaries too

I believe the actual dictionaries are baked with their host objects from the rhino plugin, but you can't see that data anywhere unless you reference the object back in grasshopper.

e-lupo, 2019-03-04:
> I believe the actual dictionaries are baked with their host objects from the rhino plugin, but you can't see that data anywhere unless you reference the object back in grasshopper.

I just checked and that is, indeed, correct. Speckle´s Get User Data can access the data after it has been baked

PaulPoinet, 2019-03-04:
Hi @e-lupo 
If you don't need to open/use Grasshopper for your purposes, another way is to write the command `GetUserText` in the Rhino command line.

PaulPoinet, 2019-03-04:
sorry please ignore the above... that's the same as checking the user data tab. user texts and user dicts are different. retrieving them through the gh interface is the best option right now unfortunately

didimitrie, 2019-06-05:
closing, as it's not really an issue (speckle uses archivable dictionaries, not user data)

closed 2019-06-05 by didimitrie