speckleworks/SpeckleRhino #276: Cannot choose Speckle Server

daviddekoning, 2019-09-11:
Step 0: 
- [x] I've  read the [contribution guidelines](https://github.com/speckleworks/SpeckleOrg/blob/master/CONTRIBUTING.md)!

#### Expected Behaviour
How things should actually work!

When a data sender or receiver is placed on the canvas, a pop up will appear that lets me log into a server of my choice.

#### Actual Behaviour
What's actually going on!

No pop up appears, and I am logged into the first server that I previously signed into (hestia in this case).

#### Affected Projects
Does this issue propagate to other dependencies or dependants? If so, list them here!

I am not sure.

#### Reproduction Steps & System Config (win, osx, web, etc.)

This is happenning on Windows 10, Rhino 6.

#### Proposed Solution (if any)

didimitrie, 2019-09-12:
Afaik, this the intended behaviour that @tecomi coded up. If you open the account manager, you have the option to set/clear your default account. If no default is present, the popup will show every time as always.

teocomi, 2019-09-13:
Currently the default account is not automatically set, but I think the previous implementation was setting it, at least when using Dynamo...