speckleworks/SpeckleRhino #301: Multiple menus are appearing in Grasshopper

jjsolly, 2020-04-16:
Step 0: 
- [ ] I've  read the [contribution guidelines](https://github.com/speckleworks/SpeckleOrg/blob/master/CONTRIBUTING.md)!

#### Expected Behaviour
How things should actually work! - Probably should only have one menu item. 

#### Actual Behaviour
What's actually going on! - Have multiple menu items

#### Affected Projects
Does this issue propagate to other dependencies or dependants? If so, list them here! - Is just annoying. 

#### Reproduction Steps & System Config (win, osx, web, etc.)
Windows 64 bit, Rhino 6 SR24, latest version Speckle (from 2 days ago)

#### Proposed Solution (if any)
[image]

teocomi, 2020-04-16:
Hello @jjsolly !
Might have been caused by some old Speckle installations, do you recall installing it previously at all or having IT installing it? Or have you maybe built and debugged it yourself?

I'd suggest trying the following solutions in order:
- uninstall Speckle form programs and features and then reinstall it
- manually delete `%localappdata%\McNeel\Rhinoceros\6.0\Plug-ins\Speckle Rhino Plugin (512d9705-6f92-49ca-a606-d6d5c1ac6aa2)\` and then reinstall Speckle

teocomi, 2020-04-16:
Sorry, closed by mistake... 🤦‍♂️

jjsolly, 2020-04-16:
Hi @teocomi it's a totally fresh installation from the other day and I did what you said already. So i don't think it should be that. But I can try again and report back. I assumed maybe there was an on-load command that can't see that the menu is already there?

teocomi, 2020-04-16:
Quite weird, aside having multiple SpeckleGrasshopper.gha on your machine I can't think on anything causing that...

PaulPoinet, 2020-04-16:
FYI @teocomi I have had the same issue quite a few times. E.g. I recently completely reset my computer and installed Speckle once, got two menus on top. For some reasons I have only one now... it's a bit unpredictable... (also not sure what to report more precisely that could help)

didimitrie, 2020-04-17:
This isssue is so annoying. The code is here: https://github.com/speckleworks/SpeckleRhino/blob/master/SpeckleGrasshopper/Loader.cs

To add a menu to the top bar, we need to run a timer function to see if gh is loaded. For some reason i never got it right, and the AddSpeckleMenu does its job more than once (or the timer elapsed function is called multiple times between when setting the `MenuHasBeenAdded` semaphore. 

I'll try and have a look next week, unless someone else picks it up.

In the meantime, please refer to this issue as the Speckle Ad Project. The faster your machine is, the more speckle menus you get.

teocomi, 2020-04-17:
^^ That's why I never had it on my machine... 🐌

daviddekoning, 2020-10-10:
the MetaHopper plugin avoids a timer by including this line in `PriorityLoad()`: 

```Instances.CanvasCreated += new Instances.CanvasCreatedEventHandler(this.RegisterNewMenuItems);```