speckleworks/SpeckleRhino #317: ApplicationID component

daviddekoning, 2020-09-02:
Step 0: 
- [x] I've  read the [contribution guidelines](https://github.com/speckleworks/SpeckleOrg/blob/master/CONTRIBUTING.md)!

### Expected Behaviour

When buildings a Speckle (1.0) model in Grasshopper, we often need to extract the ApplicationID of a Speckle Object to insert into a referencing object. We know this is solved in 2.0 with the decomposition, so we where looking for a quick way to get the applicationID from an object.

We considered:

**Expand Speckle Object**: this has two problems:

1. it is huge. It adds a lot of clutter to the GH canvas
2. The order of keys is not guaranteed. Sometimes a file is re-openned and a pipe is no longer connected to the ApplicationID component, but to some other random key.

**Query Speckle Object**: this can (currently) only access keys in the objects `properties` key.

### Proposed Solution

An applicationID component that accepts a SpeckleObject and returns it's ApplicationID.

This is very simple, but since the applicationIDs are used extensively in building SpeckleStructural models, this shortcut will be a signficant boost to adoption.

Our current prototype is here: https://github.com/arup-group/SpeckleRhino/tree/appIdComponent

#### Affected Projects

This is a grasshopper-specific intervention.

daviddekoning, 2020-09-02:
quick question regarding component icons - is there a Photoshop or GIMP template to make these? We'd like to make an icon that has a similar look to the others.

didimitrie, 2020-09-03:
Hey @daviddekoning! Looks solid and a non-threatening change, as it shouldn't bring along any other dependencies if I'm correct. 

Looking at the diff, I would maintain the old versions of rhinocommon dependencies, as there are potential changes in between those releases that we're not sure about. 

Also, I see there's a system.net dependency added to the grasshopper project file, probably that can go out? 

Re icons: i can only find the png exports :/ but i do remember that McNeel/David made all of the gh ones available here: https://developer.rhino3d.com/guides/grasshopper/grasshopper-icons/ - that should give you a much better base than my seemingly lost illustrator file!