speckleworks/SpeckleRhino #323: UserDataUtils plugin ???

jimver04, 2021-04-02:
Hi,

I have installed Speckle Server locally v2.2.5. Also installed the Rhino and Grasshoper connector. When I open from Rhino 7, the test file for Grasshoper (.gh file found in test folder it says that it needs the 'UserDataUtils' plugin) which however does not exist in Food4Rhino.

Any idea ?

Best,
Dimitrios @ PrismArch project

teocomi, 2021-04-02:
Hello Dimitrios!

I think there might have been some confusion with our old version of Speckle (v1) and the new one (v2)!
V1 is no longer maintained, so here below links to everything v2 related:

- all the repos are under another github org: https://github.com/specklesystems/
- here's the docs for v2: https://speckle.guide/
- community forum: https://speckle.community/
- website: https://speckle.systems/

Have a look at the guide and install Speckle Manager, with that you'll get our v2 connectors. I presume you are running a v2 server but a v1 connector at the moment and following a v2 tutorial :)

Hope it helps!
Matteo

jimver04, 2021-04-02:
Hi Matteo,

indeed I was looking at a Speckle v1 example. 

Nevertheless, I have managed to push commits from Rhino and Revit to speckle server (running in docker locally). 

As a second step
I am trying to execute the tutorial steps: https://speckle.guide/user/interop-gh-revit.html 
but I go very slow. 
1. I have done step "Sharing Revit project data" successfully.  
2. I have done also "Using Revit project data" in Rhino. 

I have a question, although the data is fetched in Grasshopper, how do I visualize them in 3D ?

Here I am:
[image]

Another question is how do I merge two commits ? There is only send or receive but not merge. I have experience with Github.

Best,
Dimitrios

teocomi, 2021-04-02:
Hey Dimitrios, great that you are following our tutorials! Would you mind posting your question on the [forum](https://speckle.community/c/help/8)?
That's the best place for these questions, in the meantime I'll prepare a reply :)

closed 2021-04-02 by teocomi